android/camera-samples #406: Crash when open CameraFragment

qodiriy, 2021-08-01:
E/AndroidRuntime: FATAL EXCEPTION: CameraX-core_camera_0
    Process: uz.oneid.app, PID: 2861
    java.lang.AbstractMethodError: abstract method "androidx.camera.core.CameraInfo androidx.camera.core.impl.CameraInternal.getCameraInfo()"
        at androidx.camera.core.CameraSelector.filter(CameraSelector.java:132)
        at androidx.camera.core.CameraSelector.select(CameraSelector.java:75)
        at androidx.camera.core.impl.CameraValidator.validateCameras(CameraValidator.java:81)
        at androidx.camera.core.CameraX.lambda$initAndRetryRecursively$9$CameraX(CameraX.java:613)
        at androidx.camera.core.CameraX$$ExternalSyntheticLambda10.run(Unknown Source:10)
        at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1167)
        at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:641)
        at java.lang.Thread.run(Thread.java:923)

manwithsteelnerves, 2022-04-01:
@qodiriy Did you find a solution?

MoscatoMao, 2022-07-24:
There is a high probability that it is a permission problem. Write a function to request permission.