android/camera-samples #408: I am also facing same issue in while opening camera and go to another activity and came back to camera activity in Redmi Note 8

Shivani3192, 2021-08-17:
Please follow the following instructions before filing a bug:

1. Github issues under this project are only for **sample**-related issues. If you have modified the sample code in any way, then this is probably not the appropriate place to report a bug.
1. If you are trying to report a bug related to the **CameraX library** then you can do so at the Android Issue Tracker: https://issuetracker.google.com/components/618491.
1. Include the specific sample where you found the issue, in addition to device model and specific Android build number -- which can be found under Settings > About phone.
1. Include error logs from logcat, if there are any.
1. If you can, also attach a [bug report](https://developer.android.com/studio/debug/bug-report)-- but please keep in mind that personal information can sometimes be found in those, so double check before attaching.

Shivani3192, 2021-08-17:
I am only facing this issue in Redmi Note 8, in another devices it's working fine. I am also not finding any logs as this device is not with me.

ggfan, 2021-08-25:
Thank you for the reporting the issue! Which sample you were using (or all samples have the same behavior)? Please also add the steps to replicate the issue.

ggfan, 2021-08-30:
This looks important topic, could you forward this one to also to g.co/camerax/developers (please join that group, need your cameraX experience besides this issue), thank you. It is so device dependent, I will go find a device in the mean time.

Shivani3192, 2021-09-01:
[image]
I am not able to access mentioned page @ggfan

ggfan, 2021-09-01:
@Shivani3192 the link is https://g.co/camerax/developers, the actual location is at: https://groups.google.com/a/android.com/g/camerax-developers, may you try it again?  Once you are there, clicking on the "Ask to join group" should do it. After getting in, not only share your thoughts, please also help answer some (or all) questions :-).  thank you so much!
The previous link I put there was incomplete(missed the "https://" part, then github prepends its own header), sorry.

Shivani3192, 2021-09-02:
@ggfan Thanks for the help. I have created new conversation in mentioned group please help me to resolve this issue

ggfan, 2021-09-02:
great, use that group fully: maybe beyond its limit to get best quality for your app and the lib :-). For I close this from here for now, if needed, I will pick it from that forum.

closed 2021-09-02 by ggfan