android/camera-samples #410: Camera is closed exception on takePicture()

akashsarkar188, 2021-08-25:
Hi,
I am using camera2 in my app. Generally camera works fine for most of the users but for some users when they try capture image an exception is thrown with message "Camera is closed."
While this error occurs camera is actually actively working on UI, so not sure why this error is happening.

Code I am using to capture image is : 
```
imageCapture.takePicture(outputFileOptions, ContextCompat.getMainExecutor(getContext()),
                    new ImageCapture.OnImageSavedCallback() {
                        @Override
                        public void onImageSaved(ImageCapture.OutputFileResults outputFileResults) {

                            imagePath = FileProvider
                                    .getUriForFile(getContext(), getContext().getPackageName() + ".provider", imageFile);
                            checkImageOrientation();
                        }

                        @Override
                        public void onError(ImageCaptureException error) {
                            // error.getMessage() --> Camera is closed.
                        }
                    });
```

Please let me know how to fix this.
Thank you!

ggfan, 2021-08-27:
thank you for reporting the issue! To get on this issue, may you narrow down the scope, if you could have a small app to demo it, that is great; os version, device model(s). Unless this consistently happens on most devices(Pixel device is easy to get by in our org).

akashsarkar188, 2021-09-06:
Hey @ggfan ,
Sorry for replying late, actually this is happening in our production application. So basically the app doesnt crash so I dont really have any clear logs. I am just able to get some message which says "camera is closed" in the onError callback.

Brand : Nokia 
Android version : 10
Model : Nokia 6.1 Plus

rkbhambhu, 2021-09-07:
**We are also getting this error, reported on Samsung Galaxy A52 5G, I have logs for the same. We are using CameraX stable version. The user is taking pictures continuously and it occurs in between.  Please find the logs below.
Thanks
Ram**

2021-09-01 18:22:16.79 +0530 IST:  Image file created: /storage/emulated/0/Android/data/com.insigma.iredccswb/files/Pictures/IMG_20210901_085216789.jpg
2021-09-01 18:22:19.606 +0530 IST:  Image quality is good
2021-09-01 18:22:47.489 +0530 IST:  Image file created: /storage/emulated/0/Android/data/com.insigma.iredccswb/files/Pictures/IMG_20210901_085247487.jpg
2021-09-01 18:22:50.307 +0530 IST: Image quality is good
2021-09-01 18:22:57.275 +0530 IST: Image file created: /storage/emulated/0/Android/data/com.insigma.iredccswb/files/Pictures/IMG_20210901_085257273.jpg

//LOTS OF PICTURES TAKEN AS IN THE ABOVE LOGS
//NOW ERROR OCCURED
2021-09-01 18:23:45.273 +0530 IST error CameraXActivity androidx.camera.core.ImageCaptureException: Camera is closed.\n	at androidx.camera.core.ImageCapture$ImageCaptureRequest.lambda$notifyCallbackError$1$ImageCapture$ImageCaptureRequest(ImageCapture.java:2409)\n	at androidx.camera.core.-$$Lambda$ImageCapture$ImageCaptureRequest$KlqAxzwB-08wcOFrjThjf8ncF2g.run(Unknown Source:8)\n	at android.os.Handler.handleCallback(Handler.java:883)\n	at android.os.Handler.dispatchMessage(Handler.java:100)\n	at android.os.Looper.loop(Looper.java:237)\n	at android.app.ActivityThread.main(ActivityThread.java:8167)\n	at java.lang.reflect.Method.invoke(Native Method)\n	at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:496)\n	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:1100)\nCaused by: androidx.camera.core.CameraClosedException: Camera is closed.\n	at androidx.camera.core.ImageCapture.abortImageCaptureRequests(ImageCapture.java:885)\n	at androidx.camera.core.ImageCapture.onStateDetached(ImageCapture.java:881)\n	at androidx.camera.camera2.internal.Camera2CameraImpl.notifyStateDetachedToUseCases(Camera2CameraImpl.java:732)\n	at androidx.camera.camera2.internal.Camera2CameraImpl.detachUseCases(Camera2CameraImpl.java:772)\n	at androidx.camera.core.internal.CameraUseCaseAdapter.detachUseCases(CameraUseCaseAdapter.java:292)\n	at androidx.camera.lifecycle.LifecycleCamera.onStop(LifecycleCamera.java:93)\n	at androidx.camera.lifecycle.LifecycleCamera.suspend(LifecycleCamera.java:119)\n	at androidx.camera.lifecycle.LifecycleCameraRepository.suspendUseCases(LifecycleCameraRepository.java:433)\n	at androidx.camera.lifecycle.LifecycleCameraRepository.setInactive(LifecycleCameraRepository.java:387)\n	at androidx.camera.lifecycle.LifecycleCameraRepository$LifecycleCameraRepositoryObserver.onStop(LifecycleCameraRepository.java:504)\n	at java.lang.reflect.Method.invoke(Native Method)\n	at androidx.lifecycle.ClassesInfoCache$MethodReference.invokeCallback(ClassesInfoCache.java:219)\n	at androidx.lifecycle.ClassesInfoCache$CallbackInfo.invokeMethodsForEvent(ClassesInfoCache.java:194)\n	at androidx.lifecycle.ClassesInfoCache$CallbackInfo.invokeCallbacks(ClassesInfoCache.java:185)\n	at androidx.lifecycle.ReflectiveGenericLifecycleObserver.onStateChanged(ReflectiveGenericLifecycleObserver.java:37)\n	at androidx.lifecycle.LifecycleRegistry$ObserverWithState.dispatchEvent(LifecycleRegistry.java:354)\n	at androidx.lifecycle.LifecycleRegistry.backwardPass(LifecycleRegistry.java:284)\n	at androidx.lifecycle.LifecycleRegistry.sync(LifecycleRegistry.java:302)\n	at androidx.lifecycle.LifecycleRegistry.moveToState(LifecycleRegistry.java:148)\n	at androidx.lifecycle.LifecycleRegistry.handleLifecycleEvent(LifecycleRegistry.java:134)\n	at androidx.lifecycle.ReportFragment.dispatch(ReportFragment.java:68)\n	at androidx.lifecycle.ReportFragment$LifecycleCallbacks.onActivityPreStopped(ReportFragment.java:210)\n	at android.app.Activity.dispatchActivityPreStopped(Activity.java:1345)\n	at android.app.Activity.performStop(Activity.java:8180)\n	at android.app.ActivityThread.callActivityOnStop(ActivityThread.java:5117)\n	at android.app.ActivityThread.handleSleeping(ActivityThread.java:5254)\n	at android.app.ActivityThread.access$2700(ActivityThread.java:274)\n	at android.app.ActivityThread$H.handleMessage(ActivityThread.java:2200)\n	at android.os.Handler.dispatchMessage(Handler.java:107)\n	... 5 more\n
2021-09-01 18:25:50.077 +0530 IST trace IRCamera BugfenderUtils.java:20    15197_e27e33e8-a04d-5a0f-9b42-8992ce6b4ea8 3.8.6 45 en 10 2 482  Image file created: /storage/emulated/0/Android/data/com.insigma.iredccswb/files/Pictures/IMG_20210901_085550075.jpg

bkmalkoc, 2021-10-21:
having the same issue here 

According the logs, CameraX creates 2 camera worker
```
2021-10-20 07:35:58.839 10664-10859/com.android.app D/CameraStateRegistry: Recalculating open cameras:
    Camera                                       State                 
    -------------------------------------------------------------------
    Camera@56958ac[id=1]                         UNKNOWN               
    Camera@d1c1d03[id=0]                         CLOSED                
    -------------------------------------------------------------------
    Open count: 0 (Max allowed: 1)
```

Down the road when ImageCapture tries to process the next request can't make it due to this code portion;
```

        @SuppressWarnings("WeakerAccess") /* synthetic accessor */
        void processNextRequest() {
            synchronized (mLock) {
                // Unable to issue request if there is currently a request in flight
                if (mCurrentRequest != null) {
                    return;
                }

                // Unable to issue request if the ImageReader has no available image buffer left.
                if (mOutstandingImages >= mMaxImages) {
                    Logger.w(TAG,
                            "Too many acquire images. Close image to be able to process next.");
                    return;
                }
```

Why CameraX creates 2 workers after `ImageCapture().takePicture()` being called ?

kotoMJ, 2022-12-21:
I am afraid it's not just emulator issue. I can see the happening for users in production with device Samsung Galaxy S22 Ultra. This is definitely not an emulator.

kotoMJ, 2023-01-27:
I can just add the scenario which was the cause in my particular case and how I [fixed that](https://github.com/kotoMJ/kotox-android/pull/4/commits/0352997764d0d41e02a39a8a0fc7cfd4d85a6ea6#diff-a5eea63766cddd4d9f50d7718cc543945265f684068916e56436faf2c3f0b678R63).

For my this issue happened when the app was sent to background after triggering a picture but before picture was saved. In this case camera fired ImageCaptureException. I simply try catch this exception and return null instead of photoFile.
Hope it will help others.

Spyne-Tech, 2023-02-07:
Getting same exception in 
Device Manufacturer- OPPO
OS Version- 12
Device Name- OP4EF3L1

  imageCapture1.takePicture(
            outputOptions,
            ContextCompat.getMainExecutor(requireContext()),
            object : ImageCapture.OnImageSavedCallback {
                override fun onError(exc: ImageCaptureException) {
                    **"Camera is closed"**
                }

AdarshRevankar, 2024-01-09:
For me, it just so happened that I forgot to perform `image.close()` operation over the past captured images in the following method, `override fun onCaptureSuccess(image: ImageProxy)`

Performing `takePicture` multiple times without performing  `image.close()` imageProxy might also create this issue of `androidx.camera.core.ImageCaptureException: Camera is closed.`

Hope this will be useful to someone ! Cheers 😄 ✌️

ManojKumar-Bayer, 2024-05-20:
> ContextCompat.getMainExecutor(getContext())
`imageCapture.takePicture(
            outputOptions,
            Executors.newSingleThreadExecutor(),
            object : ImageCapture.OnImageSavedCallback {
                override fun onImageSaved(outputFileResults: ImageCapture.OutputFileResults) {
                    outputFileResults.savedUri?.let { _ ->
                        if (File(filePtah).length() > VALUE_ZERO) {
                            Logger.ERROR("CAPTURE_SUCCESS", "IMAGE CAPTURED")
                            //If image written on the provided file path then directly use the path
                            onImageCapture(filePtah)
                        }else {
                            Logger.ERROR("CAPTURE_FAILED", "UNABLE TO CAPTURE IMAGE")
                        }
                    }
                }

                override fun onError(exception: ImageCaptureException) {
                    Logger.ERROR("CAPTURE_ON_ERROR", "IMAGE CAPTURING ON ERROR ${exception.message}")
                    println(exception.message)
                }
            })`
@akashsarkar188 try using separated thread for the executor(). Its worked for me.

ishanknijhawan, 2024-10-17:
Hey @akashsarkar188
Were you able to solve the issue? We're facing the same exception for our production users

RezaParivar, 2025-01-14:
I still have that problem I put a question in stackoverflow with my code in it

filipatbnp, 2026-10-02:
@MjDevCz your example does not work anymore?

> I can just add the scenario which was the cause in my particular case and how I [fixed that](https://github.com/kotoMJ/kotox-android/pull/4/commits/0352997764d0d41e02a39a8a0fc7cfd4d85a6ea6#diff-a5eea63766cddd4d9f50d7718cc543945265f684068916e56436faf2c3f0b678R63).

What happened there?