android/camera-samples #413: Some ExtensionsManager's methods are broken

MHShetty, 2021-08-30:
`ProcessCameraProvider` no longer implements `CameraProvider` but `LifecycleCameraProvider` instead. So some of the library methods, specifically `ExtensionsManager` class's some methods are broken or perhaps are no longer usable.

For example the following methods, 
```
public boolean isExtensionAvailable(@NonNull CameraProvider cameraProvider,
            @NonNull CameraSelector baseCameraSelector, @ExtensionMode.Mode int mode)

public CameraSelector getExtensionEnabledCameraSelector(@NonNull CameraProvider cameraProvider,
            @NonNull CameraSelector baseCameraSelector, @ExtensionMode.Mode int mode) 
```

But,

[image]

Could you'll please replace/overload those methods to support `ProcessCameraProvider` or `LifecycleCameraProvider`?

Thanks in advance.

ggfan, 2021-08-31:
[lifecycleCameraProvider](https://android.googlesource.com/platform/frameworks/support/+/androidx-main/camera/camera-lifecycle/src/main/java/androidx/camera/lifecycle/LifecycleCameraProvider.java#29) extends CameraProvider interface, the 2 functions should still work with a ProcessCameraProvider instance.  Something might be wrong then,  could you elaborate about the failures you saw? Thank you!

MHShetty, 2021-09-01:
Maybe there is some issue with the version of the CameraX dependencies used. 

On inspecting the source code of my library, that doesn't seem to be the case.

[image]

Here are the extensions I am using along with their version number(s) used in the project.

[image]

ggfan, 2021-09-01:
yeah, apologize for that:  the new extensions API depends camera-lifecycle:1.1.0-y, the version that is released on the same day with the camera-extensions:1.0.0-x: released on the same day, may not be the same alpha version([1.1.0-alpha08](https://developer.android.com/jetpack/androidx/releases/camera) for this case.  But we suppose to pick up the required newer version to use(not visible to you), instead of 1.0.1 configured in your build.gradle.  I will follow up on this part internally.

CharcoalChen, 2021-09-02:
Hi MHShetty,

Thanks for raising this issue. I can reproduce the issue locally. ggfan has helped to explain that the new extensions API have a dependency on core modules version 1.1.0-y. For more details, the CameraProvider interface was added and extended by the LifecycleCameraProvider interface only in camera-core 1.1.0 code base. According to your gradle dependencies setting, camera-lifecycle 1.0.1 is used. Therefore, the compiler showed the error message that ProcessCameraProvider can't be casted to a CameraProvider object. After changing the core library version to 1.1.0-alpha28, the issue is disappeared. Please have a try whether this can resolve the problem you encountered. Thanks.

MHShetty, 2021-09-02:
Is there any way I can stick to the stable version of the core libraries and still use the extensions library? (Maybe by downgrading the version of the extensions library)

Also, when could I upgrade to the latest version of the extensions library, once this issue is fixed?

Thanks a lot for prioritizing this issue :+1

CharcoalChen, 2021-09-02:
To stick using the stable version of the core libraries along with a downgraded version of camera-extensions might be possible. But it will also have some problems and potential risks such as the following: 

- You will need to use the legacy extender classes to access the extension functions. The legacy API's java doc has been removed. You will only be able to refer to [here](https://groups.google.com/a/android.com/g/camerax-developers/c/WKUERN9U2fo/m/EXaEA0KkAAAJ) to see how to use it.
- Some new devices will support CameraX Extensions by new defined vendor library interfaces. Using the legacy Extensions API can't enable effects on those devices.
- If some extensions related issues are found, you won’t be able to get the fixes from the latest ver

sion of release.
 
Based on these known problems, I tried the following camera-extensions setting with 1.0.1 core modules and the extension effects can be enabled successfully.
 
    // CameraX Extensions library
     implementation('androidx.camera:camera-extensions:1.0.0-alpha26') {
         exclude(group:"androidx.camera", module:"camera-core")
     }
 
Camera-extensions is currently in alpha stage. We’re working on moving it to the next beta stage to align it version with the 1.1.0 core modules. But it will still need to take some time. I’m not sure whether using version 1.1.0 beta core modules could be your option or not. If you still want to stick to using a stable version, it might take longer to wait for the 1.1.0 stable version.

MHShetty, 2021-09-02:
Hey @CharcoalChen. Thanks a lot for the briefing. I think I'll just skip camera extensions for a while until it gets stable or I am able to use extensions normally. If it is possible, could you'll please fix it in the upcoming alpha version? Thanks a lot 😊

MHShetty, 2021-09-02:
@ggfan Thanks for inviting @CharcoalChen and sharing about this google group. I have sent an join request. Hope they take me in soon!

MHShetty, 2021-09-18:
@ggfan @CharcoalChen 

I tried upgrading to the latest version of the core libraries in the alpha channel (i.e. 1.1.0-alpha08; 1.1.0-alpha28 doesn't exist according to the Gradle and official documentation). But the issue still remains the same. I have migrated my project to Kotlin but that doesn't make the extensions part of the library usable.

[image]
[image]

Please help me to resolve this issue asap. Thanks in advance.

CharcoalChen, 2021-09-22:
@MHShetty I tried the same gradle settings in CameraXBasic sample project by the following steps and I do not encounter the same issue in my side.

1. Download camera-samples from [here](https://github.com/android/camera-samples)
2. Open the CameraXBasic project by Android Studio
3. Update the gradle dependencies [here](https://github.com/android/camera-samples/blob/main/CameraXBasic/app/build.gradle#L71) 
    3.1 Update `camerax_version` from `1.1.0-alpha07` to `1.1.0-alpha08`
    3.2 Update `camera-view` from `1.0.0-alpha27` to `1.0.0-alpha28`
    3.3 Unmark `camera-extensions` and update its version  from `1.0.0-alpha27` to `1.0.0-alpha28`
4. Add the following code in CameraFragment.kt before calling bindToLifecycle [here](https://github.com/android/camera-samples/blob/main/CameraXBasic/app/src/main/java/com/android/example/cameraxbasic/fragments/CameraFragment.kt#L332)
```
            val extensionsManager = ExtensionsManager.getInstance(requireContext()).get()
            extensionsManager.isExtensionAvailable(cameraProvider, cameraSelector, ExtensionMode.HDR)
```

After the above changes, the dependencies setting and even the order should be the same as yours. But it works fine in my side. Could you also download the CameraXBasic sample project and try the same settings by it? You may also check Android Studio's `File -> Project Structure -> Dependencies -> app` to know which versions of the camera-core/camera2/lifecycle modules are used. The used version should be `1.1.0-alpha08`.

If the camera-core/camera2/lifecycle modules' version is still 1.0.1 in your environment, could you try to exclude the core module from camera-view and see whether it can resolve the problem?
```
 implementation('androidx.camera:camera-view:1.0.0-alpha28') {
     exclude(group:"androidx.camera", module:"camera-core")
 }
```

ggfan, 2021-09-30:
Not sure what the real problem, but sometimes Android studio might pick the cached version in your ~/.gradle/caches rather than go fetch the ones that is specified by your app. I personally just delete the things under ~/.gradle/caches (you could find the exact cache you are using by use Android studio "go to implementation" and find out directory name) -- it will waste some time(sometimes it solves my problem, I have no clue for what is happening, it seems happens more with Artic Fox version on my system) as those cache will be re-downloading. 

@MHShetty I think you connect with the team in the https://groups.google.com/a/android.com/g/camerax-developers, please continue to work things over there. We will soon add extensions to the samples here, then we will know more.

closed 2021-09-30 by ggfan