android/camera-samples #414: Use VideoCapture, ImageAnalysis and Preview use cases at the same time on CameraX

Perick95, 2021-09-03:
I've already implemented the preview, video capture and image analysis use cases for cameraX and everything works fine, I can visualize the preview, record a video or do image analysis. The problem is when I try to combine all the use cases to work together I have the error `Id : 1. May be attempting to bind too many use cases`. 
This probably because videoCapture is not already supported in cameraX. 

Is there a way to make the 3 use cases work simultaneously?

Thanks in advance.

songyating, 2022-01-10:
UseCaseGroup

songyating, 2022-01-10:
For example, Val usecasegroup = usecasegroup Builder()
.addUseCase(preview)
.addUseCase(videoCapture)
.addUseCase(imageCapture!!)
.setViewPort(viewPort)
.build()
val camera = cameraProvider. bindToLifecycle(this, cameraSelector, useCaseGroup)