android/camera-samples #423: CameraAccessException when taking photo with rear camera on emulated device running API 31

frempel, 2021-09-29:
Please follow the following instructions before filing a bug:

Model:  sdk_gphone64_x86_64
Build Number: sdk_gphone64_x86_64-userdebug 12 SPB5.210812.003 7673742 dev-keys)

When attempting to take a photo with the rear camera with the CameraXBasic sample on an emulated Pixel 5 device running Android 12, I get this message:

E/CameraCaptureSession: Session 0: Exception while stopping repeating: 
    android.hardware.camera2.CameraAccessException: CAMERA_ERROR (3): cancelRequest:507: Camera 0: Error clearing streaming request: Function not implemented (-38)
        at android.hardware.camera2.CameraManager.throwAsPublicException(CameraManager.java:1179)
        at android.hardware.camera2.impl.ICameraDeviceUserWrapper.cancelRequest(ICameraDeviceUserWrapper.java:99)
        at android.hardware.camera2.impl.CameraDeviceImpl.stopRepeating(CameraDeviceImpl.java:1287)
        at android.hardware.camera2.impl.CameraCaptureSessionImpl.close(CameraCaptureSessionImpl.java:579)
        at android.hardware.camera2.impl.CameraCaptureSessionImpl$2.onDisconnected(CameraCaptureSessionImpl.java:790)
        at android.hardware.camera2.impl.CameraDeviceImpl$7.run(CameraDeviceImpl.java:259)
        at androidx.camera.core.impl.utils.executor.SequentialExecutor$1.run(SequentialExecutor.java:109)
        at androidx.camera.core.impl.utils.executor.SequentialExecutor$QueueWorker.workOnQueue(SequentialExecutor.java:229)
        at androidx.camera.core.impl.utils.executor.SequentialExecutor$QueueWorker.run(SequentialExecutor.java:171)
        at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1167)
        at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:641)
        at java.lang.Thread.run(Thread.java:920)
     Caused by: android.os.ServiceSpecificException: cancelRequest:507: Camera 0: Error clearing streaming request: Function not implemented (-38) (code 10)
        at android.os.Parcel.createExceptionOrNull(Parcel.java:2439)
        at android.os.Parcel.createException(Parcel.java:2409)
        at android.os.Parcel.readException(Parcel.java:2392)
        at android.os.Parcel.readException(Parcel.java:2334)
        at android.hardware.camera2.ICameraDeviceUser$Stub$Proxy.cancelRequest(ICameraDeviceUser.java:749)
        at android.hardware.camera2.impl.ICameraDeviceUserWrapper.cancelRequest(ICameraDeviceUserWrapper.java:97)
        at android.hardware.camera2.impl.CameraDeviceImpl.stopRepeating(CameraDeviceImpl.java:1287) 
        at android.hardware.camera2.impl.CameraCaptureSessionImpl.close(CameraCaptureSessionImpl.java:579) 
        at android.hardware.camera2.impl.CameraCaptureSessionImpl$2.onDisconnected(CameraCaptureSessionImpl.java:790) 
        at android.hardware.camera2.impl.CameraDeviceImpl$7.run(CameraDeviceImpl.java:259) 
        at androidx.camera.core.impl.utils.executor.SequentialExecutor$1.run(SequentialExecutor.java:109) 
        at androidx.camera.core.impl.utils.executor.SequentialExecutor$QueueWorker.workOnQueue(SequentialExecutor.java:229) 
        at androidx.camera.core.impl.utils.executor.SequentialExecutor$QueueWorker.run(SequentialExecutor.java:171) 
        at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1167) 
        at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:641) 
        at java.lang.Thread.run(Thread.java:920)

Strangely, the front camera appears to work just fine.

ggfan, 2021-09-29:
I will follow up this one; but in general, I would recommend to use real devices, not the beta emulator when working on graphics and image related applications.

closed 2021-09-30 by ggfan

CHCELIA, 2023-09-03:
j'ai une erreur qui est sa 2023-09-03 22:31:46.120 12847-12920 CameraCaptureSession    com.alrujsd.remoteexpertclient       E  Session 0: Exception while stopping repeating: 
                                                                                                    android.hardware.camera2.CameraAccessException: CAMERA_ERROR (3): cancelRequest:748: Camera 0: Error clearing streaming request: Function not implemented (-38)
                                                                                                    	at android.hardware.camera2.CameraManager.throwAsPublicException(CameraManager.java:1746)
                                                                                                    	at android.hardware.camera2.impl.ICameraDeviceUserWrapper.cancelRequest(ICameraDeviceUserWrapper.java:99)
                                                                                                    	at android.hardware.camera2.impl.CameraDeviceImpl.stopRepeating(CameraDeviceImpl.java:1371)
                                                                                                    	at android.hardware.camera2.impl.CameraCaptureSessionImpl.close(CameraCaptureSessionImpl.java:579)
                                                                                                    	at android.hardware.camera2.impl.CameraCaptureSessionImpl$2.onDisconnected(CameraCaptureSessionImpl.java:790)
                                                                                                    	at android.hardware.camera2.impl.CameraDeviceImpl$7.run(CameraDeviceImpl.java:267)
                                                                                                    	at android.os.Handler.handleCallback(Handler.java:942)
                                                                                                    	at android.os.Handler.dispatchMessage(Handler.java:99)
                                                                                                    	at android.os.Looper.loopOnce(Looper.java:226)
                                                                                                    	at android.os.Looper.loop(Looper.java:313)
                                                                                                    	at android.os.HandlerThread.run(HandlerThread.java:67)
                                                                                                    Caused by: android.os.ServiceSpecificException: cancelRequest:748: Camera 0: Error clearing streaming request: Function not implemented (-38) (code 10)
                                                                                                    	at android.os.Parcel.createExceptionOrNull(Parcel.java:3037)
                                                                                                    	at android.os.Parcel.createException(Parcel.java:3007)
                                                                                                    	at android.os.Parcel.readException(Parcel.java:2990)
                                                                                                    	at android.os.Parcel.readException(Parcel.java:2932)
                                                                                                    	at android.hardware.camera2.ICameraDeviceUser$Stub$Proxy.cancelRequest(ICameraDeviceUser.java:666)
                                                                                                    	at android.hardware.camera2.impl.ICameraDeviceUserWrapper.cancelRequest(ICameraDeviceUserWrapper.java:97)
                                                                                                    	at android.hardware.camera2.impl.CameraDeviceImpl.stopRepeating(CameraDeviceImpl.java:1371) 
                                                                                                    	at android.hardware.camera2.impl.CameraCaptureSessionImpl.close(CameraCaptureSessionImpl.java:579) 
                                                                                                    	at android.hardware.camera2.impl.CameraCaptureSessionImpl$2.onDisconnected(CameraCaptureSessionImpl.java:790) 
                                                                                                    	at android.hardware.camera2.impl.CameraDeviceImpl$7.run(CameraDeviceImpl.java:267) 
                                                                                                    	at android.os.Handler.handleCallback(Handler.java:942) 
                                                                                                    	at android.os.Handler.dispatchMessage(Handler.java:99) 
                                                                                                    	at android.os.Looper.loopOnce(Looper.java:226) 
                                                                                                    	at android.os.Looper.loop(Looper.java:313) 
                                                                                                    	at android.os.HandlerThread.run(HandlerThread.java:67) 
2023-09-03 22:31:46.120 12847-12920 org.webrtc.Logging      com.alrujsd.remoteexpertclient       I  Camera2Session: Stop internal
2023-09-03 22:31:46.120 12847-12920 org.webrtc.Logging      com.alrujsd.remoteexpertclient       I  SurfaceTextureHelper: stopListening()
2023-09-03 22:31:46.121 12847-12920 org.webrtc.Logging      com.alrujsd.remoteexpertclient       I  Camera2Session: Stop done
2023-09-03 22:31:46.121 12847-12920 org.webrtc.Logging      com.alrujsd.remoteexpertclient       I  CameraCapturer: Stop capture