android/camera-samples #430: W/System.err: android.hardware.camera2.CameraAccessException: CAMERA_ERROR (3): waitUntilIdle:1313: Camera 0: Error waiting to drain: Connection timed out (-110)

arnab-kundu, 2021-10-16:
In Android 12 getting this exception while capturing image.
 `W/System.err: android.hardware.camera2.CameraAccessException: CAMERA_ERROR (3): waitUntilIdle:1313: Camera 0: Error waiting to drain: Connection timed out (-110)`

arnab-kundu, 2021-10-19:
### Steps to reproduce this issue
- I am using [Camera2Basic](https://github.com/android/camera-samples/tree/main/Camera2Basic) sample application of this repository.
 - Emulator version Android 12.
 - In emulator edit option I have selected `Show Advance Settings` -> Camera Back-> `Virtual Scene` 

 > Issue was not reproducible while I am choosing `Show Advance Settings` -> Camera Back-> `Emulated`. I am not sure its a problem of emulator or not.

naufalprakoso, 2023-03-13:
Hi @ggfan, this issue still occurs on VirtualScene and Webcam0. Does this issue come from the CameraX library or Emulator?