android/camera-samples #434: java.lang.IllegalStateException: Surface was requested when the Recorder had encountered error androidx.camera.video.internal.encoder.InvalidConfigException: android.media.MediaCodec$CodecException: Error 0xffffec77

tnt-ndricimh, 2021-11-02:
java.lang.IllegalStateException: Surface was requested when the Recorder had encountered error androidx.camera.video.internal.encoder.InvalidConfigException: android.media.MediaCodec$CodecException: Error 0xffffec77
        at androidx.camera.video.Recorder.onSurfaceRequested(Recorder.java:414)
        at androidx.camera.video.VideoCapture.createPipeline(VideoCapture.java:388)
        at androidx.camera.video.VideoCapture.resetPipeline(VideoCapture.java:435)
        at androidx.camera.video.VideoCapture.lambda$createPipeline$0$VideoCapture(VideoCapture.java:403)
        at androidx.camera.video.-$$Lambda$VideoCapture$6BBmhv2eFVUcrkw2nD0ve39oNMM.onError(Unknown Source:10)
        at androidx.camera.camera2.internal.Camera2CameraImpl.lambda$postSurfaceClosedError$13(Camera2CameraImpl.java:1153)
        at androidx.camera.camera2.internal.-$$Lambda$Camera2CameraImpl$EQgjyk5vUNeY9hNKpGr93fYzUDE.run(Unknown Source:4)
        at android.os.Handler.handleCallback(Handler.java:883)
        at android.os.Handler.dispatchMessage(Handler.java:100)
        at android.os.Looper.loop(Looper.java:237)
        at android.app.ActivityThread.main(ActivityThread.java:7860)
        at java.lang.reflect.Method.invoke(Native Method)
        at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:493)
        at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:1075)

ggfan, 2022-01-11:
Need more information:  only one capture session is supported.  do you unbind the previous one before you bind the new one?

mayowa-egbewunmi, 2022-02-14:
I am having a similar issue on Android 7.0 OS. The crash log below. This crash happens when trying to initialize camera. I am using camerax-video 1.1.0-beta01 

```
androidx.camera.video.internal.encoder.InvalidConfigException: Encoder cannot created: null
02-12 02:14:21.397 12698 12698 E AndroidRuntime: 	at androidx.camera.video.Recorder.onSurfaceRequested(Recorder.java:414)
02-12 02:14:21.397 12698 12698 E AndroidRuntime: 	at androidx.camera.video.VideoCapture.createPipeline(VideoCapture.java:388)
02-12 02:14:21.397 12698 12698 E AndroidRuntime: 	at androidx.camera.video.VideoCapture.resetPipeline(VideoCapture.java:435)
02-12 02:14:21.397 12698 12698 E AndroidRuntime: 	at androidx.camera.video.VideoCapture.lambda$createPipeline$0(VideoCapture.java:403)
02-12 02:14:21.397 12698 12698 E AndroidRuntime: 	at androidx.camera.video.VideoCapture.a(VideoCapture.java)
02-12 02:14:21.397 12698 12698 E AndroidRuntime: 	at androidx.camera.video.z.onError(Unknown Source)
02-12 02:14:21.397 12698 12698 E AndroidRuntime: 	at androidx.camera.camera2.internal.Camera2CameraImpl.lambda$postSurfaceClosedError$13(Camera2CameraImpl.java:1153)
02-12 02:14:21.397 12698 12698 E AndroidRuntime: 	at androidx.camera.camera2.internal.Camera2CameraImpl.f(Camera2CameraImpl.java)
02-12 02:14:21.397 12698 12698 E AndroidRuntime: 	at androidx.camera.camera2.internal.a0.run(Unknown Source)
02-12 02:14:21.397 12698 12698 E AndroidRuntime: 	at android.os.Handler.handleCallback(Handler.java:836)
02-12 02:14:21.397 12698 12698 E AndroidRuntime: 	at android.os.Handler.dispatchMessage(Handler.java:103)
02-12 02:14:21.397 12698 12698 E AndroidRuntime: 	at android.os.Looper.loop(Looper.java:203)
02-12 02:14:21.397 12698 12698 E AndroidRuntime: 	at android.app.ActivityThread.main(ActivityThread.java:6251)
02-12 02:14:21.397 12698 12698 E AndroidRuntime: 	at java.lang.reflect.Method.invoke(Native Method)
```

temcguir, 2022-02-14:
Thanks for the reports. This is a known issue and we are currently working on a bug fix. Will keep this thread updated on progress.

@mayowa-egbewunmi That specific issue has been addressed with [this change](https://r.android.com/1981746) which should be part of the next release.

mayowa-egbewunmi, 2022-02-14:
Thanks @temcguir for the update. I look forward to the fix

mayowa-egbewunmi, 2022-02-14:
Any timeline on when the next release will be out?

KiemDuong, 2022-02-16:
I'm looking forward to the fix too. My app has many crashes related to this issue.

**Crash log** 
`Fatal Exception: java.lang.IllegalStateException: Surface was requested when the Recorder had encountered error androidx.camera.video.internal.encoder.InvalidConfigException: android.media.MediaCodec$CodecException: Error 0xffffec77
       at androidx.camera.video.Recorder.onSurfaceRequested(Recorder.java:414)
       at androidx.camera.video.VideoCapture.createPipeline(VideoCapture.java:388)
       at androidx.camera.video.VideoCapture.resetPipeline(VideoCapture.java:435)
       at androidx.camera.video.VideoCapture.lambda$createPipeline$0$VideoCapture(VideoCapture.java:403)
       at androidx.camera.video.-$$Lambda$VideoCapture$OJ2Q9uy4uGZPV5sMTIt1w0HvLW8.onError(:10)
       at androidx.camera.camera2.internal.Camera2CameraImpl.lambda$postSurfaceClosedError$13(Camera2CameraImpl.java:1153)
       at androidx.camera.camera2.internal.-$$Lambda$Camera2CameraImpl$4vyTRU0TOPnIyEhQmEEUbqfiRFg.run(:4)
       at android.os.Handler.handleCallback(Handler.java:873)
       at android.os.Handler.dispatchMessage(Handler.java:99)
       at android.os.Looper.loop(Looper.java:214)`

tailv, 2022-02-18:
I have tried with [1.1.0-beta02](https://ci.android.com/builds/branches/aosp-androidx-main/grid?), but still encounter this issue.
@tnt-ndricimh @temcguir How about you with [1.1.0-beta02](https://ci.android.com/builds/branches/aosp-androidx-main/grid?) ?

mayowa-egbewunmi, 2022-05-16:
Hi @temcguir Do you have any update on this issue?

temcguir, 2022-05-16:
Sorry for the delay in updating this bug report. The crash should no longer occur as of `1.1.0-beta03`, though an error code might still be present in `VideoRecordEvent.Finalize`. In `1.1.0-rc01`, which was released a few days ago, the root cause of the issue was addressed, so it should no longer occur.