android/camera-samples #437: CameraX crash and functionality additions

Office-Murtaza, 2021-11-08:
Please follow the following instructions before filing a bug:

1. Github issues under this project are only for **sample**-related issues. If you have modified the sample code in any way, then this is probably not the appropriate place to report a bug.
1. If you are trying to report a bug related to the **CameraX library** then you can do so at the Android Issue Tracker: https://issuetracker.google.com/components/618491.
1. Include the specific sample where you found the issue, in addition to device model and specific Android build number -- which can be found under Settings > About phone.
1. Include error logs from logcat, if there are any.
1. If you can, also attach a [bug report](https://developer.android.com/studio/debug/bug-report)-- but please keep in mind that personal information can sometimes be found in those, so double check before attaching.

ggfan, 2021-11-08:
If there is no supported camera selectors, the sample exits ( could not continue ).  To get the resolution of the the selected QualitySelector, use [QualitySelector.getResolution()](https://developer.android.com/reference/androidx/camera/video/QualitySelector#getResolution(androidx.camera.core.CameraInfo,%20int)) after binding so you have the selected camera, and its CameraInfo. It is not supported to switching camera at run-time while capturing has started, but the feature is on the future release plan. 

Do the above answer your questions?

ggfan, 2021-11-08:
Hey @Office-Murtaza, may you kindly create a ticket [here](https://issuetracker.google.com/issues/new?component=618491&template=1257717) so we we could follow up more with the issue? thank you!

closed 2022-01-11 by ggfan