android/camera-samples #443: Camera2: Manual SurfaceView rotation

zabochen, 2021-11-10:
Hi guys. 
How can manually rotate camera2 preview, using AutoFit SurfaceView /SurfaceView from example? 
Thanks!

ggfan, 2021-11-11:
I need to learn that one(suppose to fix the multi-window/large screen problem in that class when I get chance)

zabochen, 2021-11-11:
@ggfan Ok)

1. Can't I fix the rotation issue with "AutoFitSurfaceView" now?

**Normal:**
https://drive.google.com/file/d/1Ecg0A8uO93iV4A05YoArfdcQwtCuuAQh/view?usp=sharing

**Rotation 90:**
https://drive.google.com/file/d/1qlL34fEeWWiLtf25buiTk6MNbEQ3a1rq/view?usp=sharing

2. Do I need to use the old approach with "AutoFitTextureView" to fix this issue?

https://github.com/googlearchive/android-Camera2Basic/tree/master/Application/src/main/java/com/example/android/camera2basic