android/camera-samples #450: Surface was requested when the Recorder had encountered

maliqhammad, 2021-12-09:
java.lang.IllegalStateException: Surface was requested when the Recorder had encountered error androidx.camera.video.internal.encoder.InvalidConfigException: Encoder cannot created: null
        at com.google.android.libraries.i.e.run(PG:2)
        at androidx.camera.video.Recorder.onSurfaceRequested(Recorder.java:414)
        at com.google.android.libraries.i.s.run(PG:4)
        at java.lang.Thread.run(Thread.java:764)
        at androidx.camera.video.VideoCapture.createPipeline(VideoCapture.java:388)
        at androidx.camera.video.VideoCapture.resetPipeline(VideoCapture.java:435)
        at androidx.camera.video.VideoCapture.lambda$createPipeline$0$VideoCapture(VideoCapture.java:403)
        at androidx.camera.video.VideoCapture$$ExternalSyntheticLambda0.onError(Unknown Source:10)
        at androidx.camera.camera2.internal.Camera2CameraImpl.lambda$postSurfaceClosedError$13(Camera2CameraImpl.java:1153)
     Caused by: java.lang.ClassNotFoundException: Didn't find class "android.view.contentcapture.ContentCaptureManager" on path: DexPathList[[zip file "/system/framework/org.apache.http.legacy.boot.jar", zip file "/data/app/com.google.android.googlequicksearchbox-J9dfy40HhG5R_VNj_KCg6w==/base.apk"],nativeLibraryDirectories=[/data/app/com.google.android.googlequicksearchbox-J9dfy40HhG5R_VNj_KCg6w==/lib/arm, /data/app/com.google.android.googlequicksearchbox-J9dfy40HhG5R_VNj_KCg6w==/base.apk!/lib/armeabi-v7a, /system/lib]]

evrenay, 2022-01-07:
xiaomi note 8 pro has the same error

Office-Murtaza, 2022-01-19:
QMobile s6 android version 6.0 has above error because this device not return camera quality. how to fix it.

Kathleen-Maas, 2022-04-25:
Seeing this error occur on Xiaomi, Samsung, ZTE and OPPO phones majority running android 12 and android 11

yunisrasulzade, 2022-06-11:
I`m getting crashes on Redmi Note 10 Pro and Xiaomi 11T devices

tpagram-pwc, 2022-07-01:
Observing this on Xiaomi, Motorola and Realme devices across Android 11 and Android 12

yunisrasulzade, 2022-07-04:
@tpagram-pwc, after updating to 1.2.0-alpha02 version, crashes disappeared :)

ali0094, 2022-08-10:
@yunisrasulzade  I am getting the same crash in the latest release 1.2.0-alpha04. Any solution to this ?

temcguir, 2022-08-10:
@ali0094 Are you seeing a crash or an error reported in the `Finalize` event? Are you able to post a stack trace here?

The crash should have been eliminated before 1.2, so if there is a regression we would like to know about it.

ali0094, 2022-08-11:
@temcguir I am still not able to reproduce this on debug environment.  Following crash logs were observed in Sentry  

`java.lang.IllegalStateException: Surface was requested when the Recorder had encountered error k0.g: o0.y: Encoder cannot created: null
    at androidx.camera.video.Recorder.onSurfaceRequested(SourceFile:406)
    at androidx.camera.video.VideoCapture.createPipeline(SourceFile:416)
    at androidx.camera.video.VideoCapture.resetPipeline(SourceFile:482)
    at androidx.camera.video.VideoCapture.lambda$createPipeline$0
    at androidx.camera.core.ImageCapture$$InternalSyntheticLambda$2$3ed2b896f35b77fbf132bb5abe51018b6abab991f2c5cad769ede373b29c6c94$3.onError$bridge(SourceFile:0)
    at androidx.camera.camera2.internal.Camera2CameraImpl.lambda$postSurfaceClosedError$14
    at androidx.camera.camera2.internal.Camera2CameraControlImpl$CameraControlSessionCallback$$InternalSyntheticLambda$7$8506b589ac14bfcc9c7b5ea3dc2b3c4b3714b00669d41d8948337259b7c1f8b7$0.run$bridge(SourceFile:0)
    at android.os.Handler.handleCallback(Handler.java:938)
    at android.os.Handler.dispatchMessage(Handler.java:99)
    at android.os.Looper.loop(Looper.java:223)
    at android.app.ActivityThread.main(ActivityThread.java:7945)
    at java.lang.reflect.Method.invoke(Method.java)
    at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:603)
    at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:947)`

temcguir, 2022-08-12:
@ali0094 I just double checked and the code that threw that exception was removed in version `1.2.0-alpha02`. Do you have a way of verifying which version of CameraX was being used by your app in your crash logs?

ali0094, 2022-08-12:
@temcguir I just cross verified and surprisingly PR with the updated version is not merged yet. You are right this crash is coming from version 1.1.0.

temcguir, 2022-08-18:
Thank you for double checking. I'm going to close this issue now since it is addressed in 1.2.

closed 2022-08-18 by temcguir

benedictcontawe, 2022-09-22:
Hi having this issue to.

2022-09-22 20:00:13.506 26496-26496/? E/AndroidRuntime: FATAL EXCEPTION: main
    Process: com.example.cameraapp, PID: 26496
    java.lang.IllegalStateException: Surface was requested when the Recorder had been initialized with state IDLING
        at androidx.camera.video.Recorder.onSurfaceRequested(Recorder.java:419)
        at androidx.camera.video.VideoCapture.createPipeline(VideoCapture.java:406)
        at androidx.camera.video.VideoCapture.onSuggestedResolutionUpdated(VideoCapture.java:261)
        at androidx.camera.core.UseCase.updateSuggestedResolution(UseCase.java:523)
        at androidx.camera.core.internal.CameraUseCaseAdapter.addUseCases(CameraUseCaseAdapter.java:239)
        at androidx.camera.lifecycle.LifecycleCamera.bind(LifecycleCamera.java:206)
        at androidx.camera.lifecycle.LifecycleCameraRepository.bindToLifecycleCamera(LifecycleCameraRepository.java:278)
        at androidx.camera.lifecycle.ProcessCameraProvider.bindToLifecycle(ProcessCameraProvider.java:532)
        at androidx.camera.lifecycle.ProcessCameraProvider.bindToLifecycle(ProcessCameraProvider.java:360)
        at com.example.cameraapp.VideoFragment$startCamera$1.invokeSuspend$lambda-0(VideoFragment.kt:73)
        at com.example.cameraapp.VideoFragment$startCamera$1.$r8$lambda$OQEp1medqSwsHvPfqOWvzEL7uyw(Unknown Source:0)
        at com.example.cameraapp.VideoFragment$startCamera$1$$ExternalSyntheticLambda0.run(Unknown Source:4)
        at android.os.Handler.handleCallback(Handler.java:873)
        at android.os.Handler.dispatchMessage(Handler.java:99)
        at android.os.Looper.loop(Looper.java:214)
        at android.app.ActivityThread.main(ActivityThread.java:7094)
        at java.lang.reflect.Method.invoke(Native Method)
        at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:494)
        at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:975)