android/camera-samples #452: CameraXVideo crashes on nougat with IllegalArgumentException

GarthKnight, 2021-12-10:
On emulated device android 7.1.1 app crashes when i open CameraFragment. 

2021-12-10 15:10:44.250 12221-12221/com.android.example.camerax.video E/AndroidRuntime: FATAL EXCEPTION: main
    Process: com.android.example.camerax.video, PID: 12221
    androidx.camera.core.InitializationException: java.lang.IllegalArgumentException: Log tag "CamcorderProfileResolutionQuirk" exceeds limit of 23 characters
    
        at androidx.camera.core.CameraX.lambda$initAndRetryRecursively$2$CameraX(CameraX.java:372)
        at androidx.camera.core.CameraX$$ExternalSyntheticLambda2.run(Unknown Source)
        at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1133)
        at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:607)
        at java.lang.Thread.run(Thread.java:761)
     Caused by: java.lang.IllegalArgumentException: Log tag "CamcorderProfileResolutionQuirk" exceeds limit of 23 characters
    
        at android.util.Log.isLoggable(Native Method)
        at androidx.camera.core.Logger.isLogLevelEnabled(Logger.java:76)
        at androidx.camera.core.Logger.d(Logger.java:148)
        at androidx.camera.camera2.internal.compat.quirk.CamcorderProfileResolutionQuirk.<init>(CamcorderProfileResolutionQuirk.java:89)
        at androidx.camera.camera2.internal.compat.quirk.CameraQuirks.get(CameraQuirks.java:62)
        at androidx.camera.camera2.internal.Camera2CameraInfoImpl.<init>(Camera2CameraInfoImpl.java:114)
        at androidx.camera.camera2.internal.Camera2CameraFactory.getCameraInfo(Camera2CameraFactory.java:84)
        at androidx.camera.camera2.internal.Camera2CameraFactory.getCamera(Camera2CameraFactory.java:73)
        at androidx.camera.core.impl.CameraRepository.init(CameraRepository.java:66)
        at androidx.camera.core.CameraX.lambda$initAndRetryRecursively$2$CameraX(CameraX.java:338)
        at androidx.camera.core.CameraX$$ExternalSyntheticLambda2.run(Unknown Source) 
        at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1133) 
        at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:607) 
        at java.lang.Thread.run(Thread.java:761) 
        
        Is there any way to workaround this?

tozmart, 2022-01-10:
same problem

ggfan, 2022-01-11:
looks like it is a real issue:  https://stackoverflow.com/questions/28168622/the-logging-tag-can-be-at-most-23-characters
thank you for filing the issue, it is tracked with issue tracker: https://issuetracker.google.com/issues/213989195.

closed 2022-01-11 by ggfan

ldlywt, 2022-01-14:
the same problem in  oppo 
CPH1721