android/camera-samples #463: Cannot access class 'com.google.common.util.concurrent.ListenableFuture'. Check your module classpath for missing or conflicting dependencies

taouichaimaa, 2022-01-18:
Hi, I'm trying to make the sample cameraXvideo work however , the build fails and gives this error:
````e:app/RecordingFragment.kt: (78, 52): Cannot access class 'com.google.common.util.concurrent.ListenableFuture'. Check your module classpath for missing or conflicting dependencies
e: app/RecordingFragment.kt: (78, 82): Unresolved reference. None of the following candidates is applicable because of receiver type mismatch: 
public suspend inline fun Operation.await(): Operation.State.SUCCESS defined in androidx.work
````

the line that causes this issue is : `    val cameraProvider = ProcessCameraProvider.getInstance(requireContext()).await()`
this is my build.gradle:

```
plugins {
    id 'com.android.application'
    id 'kotlin-android'
    id 'com.github.node-gradle.node'
    id("com.apollographql.apollo").version("2.5.9")

}
apply plugin: 'com.github.node-gradle.node'
apply plugin: 'kotlin-kapt'
apply plugin: 'androidx.navigation.safeargs'
def  cognitoWebClient
def apikeyPropertiesFile = rootProject.file("secrets.properties")
def apikeyProperties = new Properties()

apikeyProperties.load(new FileInputStream(apikeyPropertiesFile))
android {
    compileSdk 31

    defaultConfig {
        applicationId "beatview.android.huru"
        minSdk 21
        targetSdk 31
        versionCode 1
        versionName "1.0"
        testInstrumentationRunner "androidx.test.runner.AndroidJUnitRunner"
    }

    buildTypes {
        release {
            minifyEnabled false
            buildConfigField("String", "cognitoWebClient", apikeyProperties['cognitoWebClient'])
            proguardFiles getDefaultProguardFile('proguard-android-optimize.txt'), 'proguard-rules.pro'
        }
    }
    compileOptions {
        sourceCompatibility JavaVersion.VERSION_1_8
        targetCompatibility JavaVersion.VERSION_1_8
        coreLibraryDesugaringEnabled true

    }
    buildFeatures {
        viewBinding true
    }
    dataBinding {
        enabled = true
    }
}

dependencies {

    implementation("app.futured.donut:donut:2.2.0")
    implementation 'com.github.stfalcon:swipeable-button:0.1.0'
    implementation 'me.zhanghai.android.materialratingbar:library:1.4.0'
    implementation 'com.diogobernardino:williamchart:3.11.0'
    implementation 'com.diogobernardino.williamchart:tooltip-slider:3.11.0'
    implementation 'com.github.Akshay-Katariya:CurveBottomBar:1.0'
    implementation "com.google.android.gms:play-services-auth:19.2.0"
    implementation 'com.gauravk.bubblenavigation:bubblenavigation:1.0.7'
    implementation  'io.ak1:bubbletabbar:1.0.8'
    implementation 'com.ebanx:swipe-button:0.4.0'
    implementation 'com.dmitrymalkovich.android:progress-fab:1.6'
    implementation 'com.github.simformsolutions:SSCustomBottomNavigation:3.2'
    implementation 'com.an.otpview:otpview:0.1.0'
    implementation 'com.pawegio.kandroid:kandroid:0.8.7@aar'
    implementation 'com.google.android.material:material:1.0.0'
    implementation 'com.longer:verifycode:1.0.4'
    implementation 'com.github.imkarl:CharacterPickerView:v0.2.8'
    implementation 'com.amulyakhare:com.amulyakhare.textdrawable:1.0.1'
    implementation 'com.github.aabhasr1:OtpView:v1.1.2'
    implementation 'androidx.legacy:legacy-support-v13:1.0.0'
    implementation 'com.github.DavidProdinger:weekdays-selector:1.1.1'
    implementation "com.airbnb.android:lottie-compose:4.2.0"
    implementation "androidx.fragment:fragment-ktx:1.4.0"
    implementation 'androidx.appcompat:appcompat:1.4.1'
    implementation 'androidx.constraintlayout:constraintlayout:2.1.2'
    implementation 'androidx.annotation:annotation:1.3.0'
    implementation 'androidx.lifecycle:lifecycle-livedata-ktx:2.4.0'
    implementation 'androidx.lifecycle:lifecycle-viewmodel-ktx:2.4.0'
    implementation "androidx.lifecycle:lifecycle-compiler:2.4.0"
    implementation 'com.github.armcha:SpaceNavigationView:1.6.0'
    implementation 'androidx.legacy:legacy-support-v4:1.0.0'
    implementation 'com.github.WShaobin:VerificationCodeInputView:1.0.2'
    implementation 'androidx.navigation:navigation-fragment-ktx:2.3.5'
    //cameraX
    implementation 'androidx.camera:camera-view:1.0.0-alpha32'
    implementation 'androidx.camera:camera-video:1.1.0-alpha12'
    implementation 'androidx.camera:camera-lifecycle:1.1.0-alpha12'
    //tests
    testImplementation 'junit:junit:'
    androidTestImplementation 'androidx.test.ext:junit:1.1.3'
    androidTestImplementation 'androidx.test.espresso:espresso-core:3.4.0'
    //aws dependencies
    implementation "com.amazonaws:aws-android-sdk-core:2.36.0"
    implementation "com.amazonaws:aws-android-sdk-auth-core:2.36.0"
    implementation("com.amazonaws:aws-android-sdk-auth-userpools:2.35.0@aar") { transitive = true }
    implementation "com.amazonaws:aws-android-sdk-cognitoidentityprovider:2.36.0"
    implementation "com.amazonaws:aws-android-sdk-pinpoint:2.35.0"
    implementation 'com.amplifyframework:aws-auth-cognito:1.30.0'
    implementation 'com.amplifyframework:aws-api:1.28.3'
    implementation 'com.amplifyframework:aws-datastore:1.30.0'
    implementation 'com.amplifyframework:aws-storage-s3:1.28.3'
// Java
    implementation 'androidx.core:core:1.7.0'

    // Kotlin lang
    implementation 'androidx.core:core-ktx:1.7.0'
    implementation "org.jetbrains.kotlin:kotlin-stdlib-jdk8:1.6.10"
    implementation 'org.jetbrains.kotlinx:kotlinx-coroutines-android:1.5.2'
    implementation "androidx.concurrent:concurrent-futures-ktx:1.1.0"
    // Mobile Client for initializing the SDK
    implementation('com.amazonaws:aws-android-sdk-mobile-client:2.36.0@aar') { transitive = true }
    implementation 'org.eclipse.paho:org.eclipse.paho.client.mqttv3:1.2.0'
    implementation 'org.eclipse.paho:org.eclipse.paho.android.service:1.1.1'
    implementation 'com.amazonaws:aws-android-sdk-appsync:3.2.1'
    implementation("com.apollographql.apollo:apollo-runtime:2.5.9")
    implementation("com.apollographql.apollo:apollo-coroutines-support:2.5.9")
    implementation 'com.facebook.android:facebook-login:latest.release'
    coreLibraryDesugaring 'com.android.tools:desugar_jdk_libs:1.1.5'
    implementation 'com.github.bumptech.glide:glide:4.12.0'
    annotationProcessor 'com.github.bumptech.glide:compiler:4.12.0'
    coreLibraryDesugaring 'com.android.tools:desugar_jdk_libs:1.1.5'
    //work manager dep for notifications
    // (Java only)
    implementation("androidx.work:work-runtime:2.7.1")

    // Kotlin + coroutines
    implementation("androidx.work:work-runtime-ktx:2.7.1")
    //jetpack for permission handling
    implementation "com.google.accompanist:accompanist-permissions:0.21.1-beta"

}
```
```
`

ngengesenior, 2022-08-26:
I am also facing this issue all of a sudden. It was not happening but it just started

sjaramillo10, 2022-08-31:
I was getting this error because I was incorrectly using `implementation` for a compiler library. Basically I had to switch:
`implementation ("androidx.room:room-compiler:XXX")`
to
`kept ("androidx.room:room-compiler:XXX")`

You might have to do something similar for the line
> `implementation "androidx.lifecycle:lifecycle-compiler:2.4.0"`

tasjapr, 2022-09-22:
I fix the same problem with cameraX by adding this dependency:
`implementation "org.jetbrains.kotlinx:kotlinx-coroutines-guava:1.6.2"`

Michael-Scriven, 2024-09-19:
> I fix the same problem with cameraX by adding this dependency: `implementation "org.jetbrains.kotlinx:kotlinx-coroutines-guava:1.6.2"`

Thank you this worked for me!