android/camera-samples #467: EXIF data in `takePicture(Executor, OnImageCapturedCallback)` `onCaptureSuccess(image: ImageProxy)`

sebastinto, 2022-02-12:
Using `ExifInterface`  on the image saved to a file in `takePicture(OutputFileOptions, Executor, OnImageSavedCallback)` returns populated tags like `ImageWidth`, `ImageWidth`, `BrightnessValue`, `ISOSpeed`, etc.

However, I'd like to use `takePicture(Executor, OnImageCapturedCallback)` as it seems to capture image quite a bit faster but EXIF data is empty when using this callback.

What I'm doing in  `onCaptureSuccess(image: ImageProxy)` is as follows:

1. convert the `ImageProxy` to a `Bitmap`
2. write `Bitmap` to file

```kotlin
val bitmap = image.toBitmap()
image.close()
file.writeBitmap(bitmap, Bitmap.CompressFormat.JPEG, 100)
```

```kotklin
fun ImageProxy.toBitmap(): Bitmap {
    val planeProxy = this.planes[0]
    val buffer: ByteBuffer = planeProxy.buffer
    val bytes = ByteArray(buffer.remaining())
    buffer.get(bytes)
    return BitmapFactory.decodeByteArray(bytes, 0, bytes.size)
}
```
I would appreciate any pointer on how to get/write EXIF tags in this callback. Thanks!

sebastinto, 2022-02-12:
Well, unless I'm mistaken, it seems like this is not possible as per [this Feature Request in IssueTracker](https://issuetracker.google.com/issues/215215749).

closed 2022-02-16 by sebastinto

wenge8n, 2023-04-11:
@sebastinto 
`
val buffer = image.planes[0].buffer
                    val bytes = ByteArray(buffer.remaining()).apply { buffer.get(this) }
                    val cacheDir = externalCacheDir
                    val fileName = SimpleDateFormat("yyyy-MM-dd-HH-mm-ss-SSS", Locale.getDefault()).format(System.currentTimeMillis())
                    val outputFile = File(cacheDir, "$fileName.jpg")
                    FileOutputStream(outputFile).use { it.write(bytes) }
                    val exif = ExifInterface(outputFile)`