android/camera-samples #472: Does not work on Emulator API 31 and 32

grndvl1, 2022-02-24:
The call backs are never called for this, apparently its an issue with the camera but it same works on API 29 emulator.

`android.hardware.camera2.CameraAccessException: CAMERA_ERROR (3): cancelRequest:507: Camera 0: Error clearing streaming request: Function not implemented
`
I was getting the above error and now I am getting this too.

```
    android.hardware.camera2.CameraAccessException: CAMERA_ERROR (3): The camera device has encountered a serious error
        at android.hardware.camera2.impl.CameraDeviceImpl.checkIfCameraClosedOrInError(CameraDeviceImpl.java:2350)

```

guanshuwen, 2022-03-21:
Me too. Only on Android 11/12 emulator. Device Manager>Edit>Show Advanced Settings>Camera>Back, change it from virtual scene to emulated, and the emulator can take picture now for me.

guanshuwen, 2022-03-21:
Sometimes the log is:
```
E/CameraCaptureSession: Session 0: Exception while stopping repeating: 
    android.hardware.camera2.CameraAccessException: CAMERA_ERROR (3): The camera device has encountered a serious error
        at android.hardware.camera2.impl.CameraDeviceImpl.checkIfCameraClosedOrInError(CameraDeviceImpl.java:2350)
        at android.hardware.camera2.impl.CameraDeviceImpl.stopRepeating(CameraDeviceImpl.java:1277)
        at android.hardware.camera2.impl.CameraCaptureSessionImpl.close(CameraCaptureSessionImpl.java:579)
        at androidx.camera.camera2.internal.SynchronizedCaptureSessionBaseImpl.close(SynchronizedCaptureSessionBaseImpl.java:435)
...
```
but sometimes:
```
E/CameraCaptureSession: Session 0: Exception while stopping repeating:
android.hardware.camera2.CameraAccessException: CAMERA_ERROR (3): cancelRequest:507: Camera 0: Error clearing streaming request: Function not implemented (-38)
at android.hardware.camera2.CameraManager.throwAsPublicException(CameraManager.java:1179)
at android.hardware.camera2.impl.ICameraDeviceUserWrapper.cancelRequest(ICameraDeviceUserWrapper.java:99)
at android.hardware.camera2.impl.CameraDeviceImpl.stopRepeating(CameraDeviceImpl.java:1287)
at android.hardware.camera2.impl.CameraCaptureSessionImpl.close(CameraCaptureSessionImpl.java:579)
...
```
Don' t know why the logs are different with the same apk.

Also happends in[camera-samples](https://github.com/android/camera-samples/issues/423)