android/camera-samples #476: Using video and image capture use-cases together causing front camera preview lag on HUAWEI P40 PRO

ali0094, 2022-03-22:
CameraX video capture use case preview is causing lag when switched to the front-facing camera on HUAWEI P40 PRO with Android OS 12.0
Issue has been observed in official CameraXVideo sample and in the latest release as well.

temcguir, 2022-03-22:
Thank you for filing the issue. Would you please provide a bit more information?

* What `Quality` setting are you using for the `Recorder`? Does the lag occur with different `Quality`?
* Do you observe the lag before, during or after starting a recording?
* Does the recorded video also appear laggy when it is played back?

I have created an issue on the issue tracker to follow this bug: https://issuetracker.google.com/226136997
Feel free to respond here or on that bug.

ali0094, 2022-03-23:
Hi @temcguir, I have tried following solutions so far

1. Setting quality highest (UHD) and lowest (SD) , issue still persists.
2. Lag is coming before and after recording both and back camera is working fine and as soon as I am switching to front camera, it is laging.
3. Yes video is also laggy

Here is my implementation :

 `QualitySelector.fromOrderedList(
ArrayList(filteredQualities), 
FallbackStrategy.lowerQualityOrHigherThan(Quality.SD)
)`

`val camera = cameraProvider.bindToLifecycle(
                    this,
                    cameraSelector,
                    preview,
                    imageCapture,
                    videoCapture
                )`

temcguir, 2022-03-24:
Thanks for the follow-up. I'm trying to get one of these devices and will try to reproduce once I can.

tri-cloud, 2022-03-25:
Anyone know what www. file magic.com is I keep seeing it in my chrome
history