android/camera-samples #485: VideoCapture throwing pending recording error after camera switch

chief10x, 2022-05-23:
Hi
I'm using camera x version 1.1.0-rc01, the issue has been persistent since beta versions of this release.
I can record easy without any issues on default camera state, but as soon as I switch the camera record fails.

`

cameraProviderFuture.addListener({

            val cameraProvider = cameraProviderFuture.get()

            val cameraCapabilities = CameraUtils.getCameraCapabilities(cameraProvider, lifecycleOwner)

            val cameraSelector =
                cameraCapabilities[viewModel.lensFacing.value % cameraCapabilities.size].camSelector
            val quality = cameraCapabilities[viewModel.lensFacing.value].qualities[qualityIndex]
            val qualitySelector = QualitySelector.from(quality)

            val recorder = Recorder.Builder()
                .setQualitySelector(qualitySelector)
                .setExecutor(cameraExecutor)
                .build()

            // Video Capture.camera is null
            videoCapture = VideoCapture.withOutput(recorder)

            val preview = Preview.Builder().build().also {
                it.setSurfaceProvider(previewView.surfaceProvider)
            }

            try {
                cameraProvider.unbindAll()
                cameraProvider.bindToLifecycle(
                    lifecycleOwner,
                    cameraSelector,
                    videoCapture,
                    preview
                )
            } catch (e: Exception) {
                e.printStackTrace()
            }

            preview.setSurfaceProvider(previewView.surfaceProvider)

        }, cameraExecutor)
`

chief10x, 2022-05-23:
This is the code to rebind the camera, I'm using compose btw

temcguir, 2022-05-23:
Can you post the error that is being thrown, along with stack trace if possible?

chief10x, 2022-05-24:
java.lang.RuntimeException: Recording was stopped before any data could be produced.
W/System.err:     at androidx.camera.video.Recorder.stop(Recorder.java:856)
W/System.err:     at androidx.camera.video.Recording.stop(Recording.java:164)
W/System.err:     at com.brief.snacks.ui.create.components.capture.CameraComponentKt$CameraComponent$1$2.emit(CameraComponent.kt:101)
W/System.err:     at com.brief.snacks.ui.create.components.capture.CameraComponentKt$CameraComponent$1$2.emit(CameraComponent.kt:74)
W/System.err:     at kotlinx.coroutines.flow.internal.SafeCollectorKt$emitFun$1.invoke(SafeCollector.kt:15)
W/System.err:     at kotlinx.coroutines.flow.internal.SafeCollectorKt$emitFun$1.invoke(SafeCollector.kt:15)
W/System.err:     at kotlinx.coroutines.flow.internal.SafeCollector.emit(SafeCollector.kt:87)
W/System.err:     at kotlinx.coroutines.flow.internal.SafeCollector.emit(SafeCollector.kt:66)
W/System.err:     at androidx.compose.runtime.SnapshotStateKt__SnapshotFlowKt$snapshotFlow$1.invokeSuspend(SnapshotFlow.kt:160)
W/System.err:     at kotlin.coroutines.jvm.internal.BaseContinuationImpl.resumeWith(ContinuationImpl.kt:33)
W/System.err:     at kotlinx.coroutines.DispatchedTask.run(DispatchedTask.kt:106)
W/System.err:     at androidx.compose.ui.platform.AndroidUiDispatcher.performTrampolineDispatch(AndroidUiDispatcher.android.kt:81)
W/System.err:     at androidx.compose.ui.platform.AndroidUiDispatcher.access$performTrampolineDispatch(AndroidUiDispatcher.android.kt:41)
W/System.err:     at androidx.compose.ui.platform.AndroidUiDispatcher$dispatchCallback$1.run(AndroidUiDispatcher.android.kt:57)
W/System.err:     at android.os.Handler.handleCallback(Handler.java:938)
W/System.err:     at android.os.Handler.dispatchMessage(Handler.java:99)
W/System.err:     at android.os.Looper.loopOnce(Looper.java:201)
W/System.err:     at android.os.Looper.loop(Looper.java:288)
W/System.err:     at android.app.ActivityThread.main(ActivityThread.java:7842)
W/System.err:     at java.lang.reflect.Method.invoke(Native Method)
W/System.err:     at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:548)
W/System.err:     at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:1003)

syslogic, 2022-06-03:
This appears to be rather for the CameraX [issue tracker](https://issuetracker.google.com/issues?q=componentid:618491), than `camera-samples`.<br/>The first one issue I've randomly clicked turned up this [`ComposeUiFragment.kt`](https://cs.android.com/androidx/platform/frameworks/support/+/androidx-main:camera/integration-tests/viewtestapp/src/main/java/androidx/camera/integration/view/ComposeUiFragment.kt)