android/camera-samples #496: CameraX providing Distorted bitmap when saving ImageProxy to bitmap.

HemanParbhakar, 2022-07-19:
When trying to save the bitmap from ImageProxy it save the bitmap as distorted but only in case of FrontCamera. we have tested on multiple devices the issues only appears in OnePlus6T on camera2 it was working properly. If we remove the imageCapture from bind to lifecycle then it start returning proper bitmap in OnepPlus6T devices as well.
Changes done instead of **setAspectRatio** **setTargetResolution** is used where size is calculated using 
` val metrics = windowManager.getCurrentWindowMetrics().bounds`
ImageAnalzyer now looks like this. 
              `  imageAnalyzer = ImageAnalysis.Builder()
                .setTargetResolution(maxSize)
                .setTargetRotation(rotation)
                .setOutputImageFormat(ImageAnalysis.OUTPUT_IMAGE_FORMAT_RGBA_8888)
                .setBackpressureStrategy(ImageAnalysis.STRATEGY_KEEP_ONLY_LATEST)
                .build()
                .also {
                    it.setAnalyzer(cameraExecutor, this)
                }`
Bitmap is intialized 
           `bitmap = Bitmap.createBitmap(imageSize.width, imageSize.height, Bitmap.Config.ARGB_8888);`
Bitmap save is done in override analyze function
          ` bitmap!!.copyPixelsFromBuffer(image.image!!.planes[0].buffer);
            val bos = ByteArrayOutputStream()
            cameraBitmap!!.compress(Bitmap.CompressFormat.JPEG, 100, bos)
            val bitmapdata = bos.toByteArray()`

Its happening on OnePlus 6T(A6010)
Android Version : 11
Oxygen OS Version : 11.1.2.2
Build Number : ONEPLUS A6010_41_21125

kareemalkoul, 2022-07-21:
if the bitmap show like this photo

[image]

just throwaway the old way converter and use **this way** ,it's work with custom size seletced

```
val myByteArray: ByteArray? = ImageUtil.imageToJpegByteArray(imageProxy)
val inputStream = ByteArrayInputStream(myByteArray)
val bitmap = BitmapFactory.decodeStream(inputStream) 
```

bbfeechen, 2022-07-21:
It could be device-specific reason that rowStride doesn't equal to imageWidth on OnepPlus6T.
So instead of using copyPixelsFromBuffer (doesn't consider pixelStride or rowStride), you could try createBitmap/setPixels or other Bitmap APIs.

e.g. Try this code snippt to see if it works
Bitmap targetBitmap = Bitmap.createBitmap(
        image.getWidth(),
        image.getHeight(),
        Bitmap.Config.ARGB_8888);

ByteBuffer byteBuffer =
        image.getImage().getPlanes()[0].getBuffer();
int index = 0;
for (int i = 0; i < image.getHeight(); i++) {
    for (int j = 0; j < image.getWidth(); j++) {
        int R = byteBuffer.get(index);
        int G = byteBuffer.get(index + 1);
        int B = byteBuffer.get(index + 2);
        int A = byteBuffer.get(index + 3);

        int color =
                (A & 0xff) << 24 | (R & 0xff) << 16 | (G & 0xff) << 8 | (B & 0xff);
        targetBitmap.setPixel(j,  i, color);
        index += 4;
    }
}

FrancescoSantagati, 2022-10-05:
You could use copyPixelsFromBuffer using plane properties instead of width:

```kotlin
fun Image.toBitmap(): Bitmap {
    val plane: Image.Plane = planes[0]
    val targetBitmap = Bitmap.createBitmap(plane.rowStride / plane.pixelStride, height, Bitmap.Config.ARGB_8888)
    targetBitmap.copyPixelsFromBuffer(plane.buffer)
    return targetBitmap
}
```