android/camera-samples #505: Copyright infringement by stripping the author's name

syslogic, 2022-08-17:
It appears to me as my authorship had been infringed upon, which I find highly unprofessional.

> Whenever a non-Googler wants to submit a patch to a Google open source project, they must first sign a Contributor License Agreement (CLA). **This allows the contributor to retain their ownership in the code submitted** while granting Google the necessary legal rights to use that contribution

https://github.com/android/camera-samples/pull/489#event-7208172009

Got anything to say to this unserious practice?

donovanfm, 2022-08-17:
Hi, syslogic. Are you saying that you are seeing changes you authored on the android:main branch?

I can't see your changes anymore in that PR, except for the one diff I commented on. If you look at [that line](https://github.com/android/camera-samples/blob/main/Camera2Basic/app/build.gradle#L34) in our main branch, you'll see that it's still the original line without your change.

syslogic, 2022-08-17:
The code of the PR had been copied in pristine form ...this is what version control says.
Now I just want to cancel this CLA, because I don't agree to such unserious methods ...
but there isn't a delete button: https://cla.developers.google.com/clas

I'm also planning to migrate most to AWS or Huawei Cloud because of this.

donovanfm, 2022-08-18:
It sounds like you discovered this was a problem with version control and some merge conflicts. I'm going to close this issue since there's no problem with the repo itself.

syslogic, 2022-08-18:
@donovanfm The problem is that you didn't merge the PR and therefore caused that mess on my side.

closed 2022-08-18 by syslogic