android/camera-samples #510: Error in Camera2Extensions

AstroTej, 2022-09-28:
Once it is run on Emulator or n phone, after taking the camera permission, it just shows a blank screen. Following is the content in the emulator:

09/28 09:43:00: Launching 'app' on Pixel 5 API 33.
Install successfully finished in 446 ms.
$ adb shell am start -n "com.android.example.camera2.extensions/com.example.android.camera2.extensions.CameraActivity" -a android.intent.action.MAIN -c android.intent.category.LAUNCHER
Connected to process 5275 on device 'Pixel_5_API_33 [emulator-5554]'.
Capturing and displaying logcat messages from application. This behavior can be disabled in the "Logcat output" section of the "Debugger" settings page.
I/era2.extensions: Late-enabling -Xcheck:jni
W/era2.extensions: Unexpected CPU variant for x86: x86_64.
    Known variants: atom, sandybridge, silvermont, kabylake, default
W/re-initialized>: type=1400 audit(0.0:19): avc: granted { execute } for path="/data/data/com.android.example.camera2.extensions/code_cache/startup_agents/4ba71210-agent.so" dev="dm-33" ino=44272 scontext=u:r:untrusted_app_30:s0:c157,c256,c512,c768 tcontext=u:object_r:app_data_file:s0:c157,c256,c512,c768 tclass=file app=com.android.example.camera2.extensions
V/studio.deploy: Startup agent attached to VM
V/studio.deploy: No existing instrumentation found. Loading instrumentation from instruments-1f2f27f8.jar
W/era2.extensions: DexFile /data/data/com.android.example.camera2.extensions/code_cache/.studio/instruments-1f2f27f8.jar is in boot class path but is not in a known location
V/studio.deploy: ModifyParameterTransform failed: Landroid/app/ApplicationLoaders;
W/era2.extensions: Current dex file has more than one class in it. Calling RetransformClasses on this class might fail if no transformations are applied to it!
W/era2.extensions: Current dex file has more than one class in it. Calling RetransformClasses on this class might fail if no transformations are applied to it!
W/era2.extensions: Redefining intrinsic method java.lang.Thread java.lang.Thread.currentThread(). This may cause the unexpected use of the original definition of java.lang.Thread java.lang.Thread.currentThread()in methods that have already been compiled.
W/era2.extensions: Redefining intrinsic method boolean java.lang.Thread.interrupted(). This may cause the unexpected use of the original definition of boolean java.lang.Thread.interrupted()in methods that have already been compiled.
W/era2.extensions: Current dex file has more than one class in it. Calling RetransformClasses on this class might fail if no transformations are applied to it!
W/era2.extensions: Current dex file has more than one class in it. Calling RetransformClasses on this class might fail if no transformations are applied to it!
W/era2.extensions: Current dex file has more than one class in it. Calling RetransformClasses on this class might fail if no transformations are applied to it!
D/CompatibilityChangeReporter: Compat change id reported: 171979766; UID 10157; state: ENABLED
W/ziparchive: Unable to open '/data/app/~~UCIMFVszp0CZ_ykZwB3NOA==/com.android.example.camera2.extensions-tCDi74yj9YikRM3XpRzwSA==/base.dm': No such file or directory
W/ziparchive: Unable to open '/data/app/~~UCIMFVszp0CZ_ykZwB3NOA==/com.android.example.camera2.extensions-tCDi74yj9YikRM3XpRzwSA==/base.dm': No such file or directory
V/GraphicsEnvironment: ANGLE Developer option for 'com.android.example.camera2.extensions' set to: 'default'
V/GraphicsEnvironment: ANGLE GameManagerService for com.android.example.camera2.extensions: false
V/GraphicsEnvironment: Neither updatable production driver nor prerelease driver is supported.
D/NetworkSecurityConfig: No Network Security Config specified, using platform default
D/NetworkSecurityConfig: No Network Security Config specified, using platform default
D/libEGL: loaded /vendor/lib64/egl/libEGL_emulation.so
D/libEGL: loaded /vendor/lib64/egl/libGLESv1_CM_emulation.so
D/libEGL: loaded /vendor/lib64/egl/libGLESv2_emulation.so
W/era2.extensions: Accessing hidden method Landroid/view/View;->computeFitSystemWindows(Landroid/graphics/Rect;Landroid/graphics/Rect;)Z (unsupported, reflection, allowed)
W/era2.extensions: Accessing hidden method Landroid/view/ViewGroup;->makeOptionalFitsSystemWindows()V (unsupported, reflection, allowed)
I/CameraManagerGlobal: Connecting to camera service
D/HostConnection: createUnique: call
D/HostConnection: HostConnection::get() New Host Connection established 0x77a5768c9a10, tid 5308
D/HostConnection: HostComposition ext ANDROID_EMU_CHECKSUM_HELPER_v1 ANDROID_EMU_native_sync_v2 ANDROID_EMU_native_sync_v3 ANDROID_EMU_native_sync_v4 ANDROID_EMU_dma_v1 ANDROID_EMU_direct_mem ANDROID_EMU_host_composition_v1 ANDROID_EMU_host_composition_v2 ANDROID_EMU_YUV_Cache ANDROID_EMU_has_shared_slots_host_memory_allocator ANDROID_EMU_sync_buffer_data ANDROID_EMU_read_color_buffer_dma ANDROID_EMU_hwc_multi_configs GL_OES_vertex_array_object GL_KHR_texture_compression_astc_ldr ANDROID_EMU_host_side_tracing ANDROID_EMU_gles_max_version_2 
W/OpenGLRenderer: Failed to choose config with EGL_SWAP_BEHAVIOR_PRESERVED, retrying without...
W/OpenGLRenderer: Failed to initialize 101010-2 format, error = EGL_SUCCESS
D/EGL_emulation: eglCreateContext: 0x77a5768ca190: maj 2 min 0 rcv 2
D/EGL_emulation: eglMakeCurrent: 0x77a5768ca190: ver 2 0 (tinfo 0x77a796efa080) (first time)
I/Gralloc4: mapper 4.x is not supported
D/HostConnection: createUnique: call
D/HostConnection: HostConnection::get() New Host Connection established 0x77a5768ce390, tid 5308
D/goldfish-address-space: allocate: Ask for block of size 0x100
D/goldfish-address-space: allocate: ioctl allocate returned offset 0x3f3ffe000 size 0x2000
W/Gralloc4: allocator 4.x is not supported
D/HostConnection: HostComposition ext ANDROID_EMU_CHECKSUM_HELPER_v1 ANDROID_EMU_native_sync_v2 ANDROID_EMU_native_sync_v3 ANDROID_EMU_native_sync_v4 ANDROID_EMU_dma_v1 ANDROID_EMU_direct_mem ANDROID_EMU_host_composition_v1 ANDROID_EMU_host_composition_v2 ANDROID_EMU_YUV_Cache ANDROID_EMU_has_shared_slots_host_memory_allocator ANDROID_EMU_sync_buffer_data ANDROID_EMU_read_color_buffer_dma ANDROID_EMU_hwc_multi_configs GL_OES_vertex_array_object GL_KHR_texture_compression_astc_ldr ANDROID_EMU_host_side_tracing ANDROID_EMU_gles_max_version_2 
W/Parcel: Expecting binder but got null!
W/System: A resource failed to call close.

hieuvts, 2022-11-17:
### SelectorFragment.kt
```
private fun enumerateExtensionCameras(cameraManager: CameraManager): List<CameraInfo> {
            ...................
              if (capabilities.contains(CameraCharacteristics
                                          .REQUEST_AVAILABLE_CAPABILITIES_BACKWARD_COMPATIBLE) &&
                extensionCharacteristics.supportedExtensions.isNotEmpty()) {
                availableCameras.add(CameraInfo("$orientation ($id)", id))
              }
            }

            return availableCameras
        }
```
Maybe `extensionCharacteristics.supportedExtensions` return false.

naimmolla916, 2022-12-05:
private class YourImageAnalyzer : ImageAnalysis.Analyzer {

    override fun analyze(imageProxy: ImageProxy) {
        val mediaImage = imageProxy.image
        if (mediaImage != null) {
            val image = InputImage.fromMediaImage(mediaImage, imageProxy.imageInfo.rotationDegrees)
            // Pass image to an ML Kit Vision API
            // ...
        }
    }
}val options = BarcodeScannerOptions.Builder()
        .setBarcodeFormats(
                Barcode.FORMAT_QR_CODE,
                Barcode.FORMAT_AZTEC)
        .build()<application ...>
      ...
      <meta-data
          android:name="com.google.mlkit.vision.DEPENDENCIES"
          android:value="barcode" >
      
</application>dependencies {
  // ...
  // Use this dependency to use the dynamically downloaded model in Google Play Services
  implementation 'com.google.android.gms:play-services-mlkit-barcode-scanning:18.1.0'
}