android/camera-samples #521: Camera2Video records videos without sound

TimSet0815, 2022-12-15:
Hello,
I noticed that the sample Camera2Video does only record videos without sound. I would have expected that the videos are with sound because the permission for audio recording is set. 
Maybe it is only an issue with my Samsung Galaxy S22 running Andriod 13. Has anyone experienced the same issue?
Greetings,
Tim

YoussefHachicha, 2023-05-13:
Hey tim how did you get Camera2Video to work for you, it dosent work for me do you still have the old version, can you provide me with it cause they apparantly made some changes to it , but now it have some issues

borgerag, 2023-06-01:
Apologies for the very late reply - when moving to a different encoding method, audio recording was simply removed because it was not core to the demo. The goal of these sample apps are to demonstrate use of the Camera2 API. If there's demand, I can add it back in.

rohaitas, 2023-06-27:
@borgerag   Please guide us. How can we record audio alongside the video in camera2video.

borgerag, 2023-07-31:
This has now been addressed. A simplified MediaRecorder path has been added with the mic enabled.

madebymozart, 2023-07-31:
Thanks @borgerag !