android/camera-samples #561: Where are the "Dual concurrent camera" CameraX Examples?

rbarbish, 2023-09-28:
As of [CameraX 1.3](https://android-developers.googleblog.com/2023/06/camerax-13-is-now-in-beta.html) (June 2023) it looks like CameraX added support for simultaneous front facing and back facing camera recordings, aka PIP (Picture-In-Picture) video recordings. Apple has had this since 2019 via [AVCaptureMultiCamSession](https://developer.apple.com/documentation/avfoundation/avcapturemulticamsession) so I thought there would be a ton more buzz and excitement about this functionality being added to CameraX, but I can't find anything other than a single blogpost. Where are the examples/samples for how to use this new functionality?!

kailianc, 2023-09-28:
This is the sample app to show a front + back camera previews in Picture-in-Picture layout or Side-by-Side layout: https://android.googlesource.com/platform/frameworks/support/+/refs/heads/androidx-main/camera/integration-tests/coretestapp/src/main/java/androidx/camera/integration/core/ConcurrentCameraActivity.java#239

In v1, we are only supporting two streams output. To support VideoCapture as one stream, you need to write your own compositing layer (one option is to use OpenGL ES) and output to MediaCodec. If you need this feature, please file a feature request to us.

rbarbish, 2023-09-28:
@kailianc Thanks for the response.

I'm getting `plugin [id: 'AndroidXPlugin'] was not found in any of the following sources` error when I try to build the coretestapp. I see it is included elsewhere in the repository so currently trying to pull it in and get it building and running on my test Pixel 5A.

Just to clarify, there are no known examples or sample projects of a video recording application with the back-facing camera filling the screen and the front-facing camera as a PIP (thumbnail) in the bottom corner (or vice versa) with the ability to save to the phone's camera roll?

If not, where might a file a feature request? Thank you

kailianc, 2023-09-28:
Yes, you could file a feature request to https://issuetracker.google.com/issues/new?component=618491&template=1257717.
We will prioritize the feature in the next version of this feature.

To mention, not all devices will support concurrent dual camera, you could check the capability via https://developer.android.com/reference/android/content/pm/PackageManager#FEATURE_CAMERA_CONCURRENT

Let me know if you have another questions.

rbarbish, 2023-09-28:
@kailianc I created https://issuetracker.google.com/issues/302562961

kailianc, 2023-09-28:
Thanks for the feedback.