android/camera-samples #564: How to use camerax 1.3.0-alpha04 to open two cameras

lodgkk, 2023-11-14:
How to use camerax 1.3.0-alpha04 to open two cameras

kailianc, 2023-11-14:
Do you want to open front + rear cameras at the same time?
https://developer.android.com/reference/androidx/camera/lifecycle/ProcessCameraProvider#getAvailableConcurrentCameraInfos%28%29

Before that, you need to check this flag to see if device is supporting this feature: https://developer.android.com/reference/android/content/pm/PackageManager#FEATURE_CAMERA_CONCURRENT

lodgkk, 2023-11-17:
I get an error Concurrent camera is not supported on the device , but I can open two cameras at the same time when using camera1. How can I solve this?

lodgkk, 2023-11-17:
> Do you want to open front + rear cameras at the same time? https://developer.android.com/reference/androidx/camera/lifecycle/ProcessCameraProvider#getAvailableConcurrentCameraInfos%28%29
>
> Before [...]

I get an error Concurrent camera is not supported on the device , but I can open two cameras at the same time when using camera1. How can I solve this?

kailianc, 2023-11-17:
Are you opening the front and rear cameras? It's possible that some OEMs don't advertise this feature but you are still be able to try, but the stability is not guaranteed. From framework's perspective, we treat this flag as the source of truth.

lodgkk, 2023-11-20:
> Are you opening the front and rear cameras? It's possible that some OEMs don't advertise this feature but you are still be able to try, but the stability is not guaranteed. From framework's perspective, we treat this flag as the source of truth.

My project is an access control project. When using camera1, it is possible to simultaneously turn on the rgb camera and infrared camera. When using camerax, I have tried turning on the rgb camera and infrared camera separately, and both worked without issue.

kailianc, 2023-11-20:
CameraX Concurrent camera API only refers to opening front + rear camera on the device at the same time. Opening rgb camera and infrared camera is different. If you want to open front + rear camera, you could use concurrent camera API.

iamwilliamli, 2023-12-03:
@kailianc Is it possible to turn on two rear camera for video recording at same time?