android/camera-samples #569: Cannot build Camera2Basic

stevebroshar, 2024-01-16:
Please follow the following instructions before filing a bug:

1. Github issues under this project are only for **sample**-related issues. If you have modified the sample code in any way, then this is probably not the appropriate place to report a bug.
1. If you are trying to report a bug related to the **CameraX library** then you can do so at the Android Issue Tracker: https://issuetracker.google.com/components/618491.
1. Include the specific sample where you found the issue, in addition to device model and specific Android build number -- which can be found under Settings > About phone.
1. Include error logs from logcat, if there are any.
1. If you can, also attach a [bug report](https://developer.android.com/studio/debug/bug-report)-- but please keep in mind that personal information can sometimes be found in those, so double check before attaching.

Summary: I cannot build Camera2Basic. I would hope and assume samples would be easy to build. But I get one error after another. And I find no instructions on how to get it building. Not a very useful sample if I can't build it.

Sample: Camera2Basic
Target: Pixel 7 API 34 (also tried API 30)
Using latest from https://github.com/android/camera-samples/tree/main/Camera2Basic
Open in Android studio 2023.1.1
Fails to build with logged message:
	Unsupported Java. 
	Your build is currently configured to use Java 17.0.7 and Gradle 6.7.1.
	Possible solution:
	 - Upgrade Gradle wrapper to 7.2 version and re-import the project
Clicked link to upgrade gradle; seems to have succeeded
Build still failing; there is no log output, but almost immediately, I see popup: 
	Unable to find Gradle tasks to build: [:]. Build mode: ASSEMBLE. Tests: All.
A few seconds later, clicked build again (no other changes) and then it started building!!!
Fails with LOTS OF INFO (too much!). There is a hierarchy of errors logged and two popups! What is the important stuff????? 
[image]
 
The longest log message is as follows:
> Task :utils:compileDebugKotlin
Compilation with Kotlin compile daemon was not successful
java.rmi.ServerError: Error occurred in server thread; nested exception is: 
	java.lang.NoClassDefFoundError: Could not initialize class org.jetbrains.kotlin.com.intellij.util.io.FileChannelUtil
	at java.rmi/sun.rmi.server.UnicastServerRef.dispatch(UnicastServerRef.java:389)
	at java.rmi/sun.rmi.transport.Transport$1.run(Transport.java:200)
	at java.rmi/sun.rmi.transport.Transport$1.run(Transport.java:197)
	at java.base/java.security.AccessController.doPrivileged(AccessController.java:712)
	at java.rmi/sun.rmi.transport.Transport.serviceCall(Transport.java:196)
	at java.rmi/sun.rmi.transport.tcp.TCPTransport.handleMessages(TCPTransport.java:587)
	at java.rmi/sun.rmi.transport.tcp.TCPTransport$ConnectionHandler.run0(TCPTransport.java:828)
	at java.rmi/sun.rmi.transport.tcp.TCPTransport$ConnectionHandler.lambda$run$0(TCPTransport.java:705)
	at java.base/java.security.AccessController.doPrivileged(AccessController.java:399)
	at java.rmi/sun.rmi.transport.tcp.TCPTransport$ConnectionHandler.run(TCPTransport.java:704)
	at java.base/java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1136)
	at java.base/java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:635)
	at java.base/java.lang.Thread.run(Thread.java:833)
	at java.rmi/sun.rmi.transport.StreamRemoteCall.exceptionReceivedFromServer(StreamRemoteCall.java:304)
	at java.rmi/sun.rmi.transport.StreamRemoteCall.executeCall(StreamRemoteCall.java:280)
	at java.rmi/sun.rmi.server.UnicastRef.invoke(UnicastRef.java:165)
	at java.rmi/java.rmi.server.RemoteObjectInvocationHandler.invokeRemoteMethod(RemoteObjectInvocationHandler.java:215)
	at java.rmi/java.rmi.server.RemoteObjectInvocationHandler.invoke(RemoteObjectInvocationHandler.java:160)
	at jdk.proxy4/jdk.proxy4.$Proxy207.compile(Unknown Source)
	at org.jetbrains.kotlin.compilerRunner.GradleKotlinCompilerWork.incrementalCompilationWithDaemon(GradleKotlinCompilerWork.kt:293)
	at org.jetbrains.kotlin.compilerRunner.GradleKotlinCompilerWork.compileWithDaemon(GradleKotlinCompilerWork.kt:216)
	at org.jetbrains.kotlin.compilerRunner.GradleKotlinCompilerWork.compileWithDaemonOrFallbackImpl(GradleKotlinCompilerWork.kt:153)
	at org.jetbrains.kotlin.compilerRunner.GradleKotlinCompilerWork.run(GradleKotlinCompilerWork.kt:132)
	at org.jetbrains.kotlin.compilerRunner.GradleCompilerRunner.runCompilerAsync(GradleKotlinCompilerRunner.kt:177)
	at org.jetbrains.kotlin.compilerRunner.GradleCompilerRunner.runCompilerAsync(GradleKotlinCompilerRunner.kt:172)
	at org.jetbrains.kotlin.compilerRunner.GradleCompilerRunner.runJvmCompilerAsync(GradleKotlinCompilerRunner.kt:87)
	at org.jetbrains.kotlin.gradle.tasks.KotlinCompile.callCompilerAsync$kotlin_gradle_plugin(Tasks.kt:480)
	at org.jetbrains.kotlin.gradle.tasks.KotlinCompile.callCompilerAsync$kotlin_gradle_plugin(Tasks.kt:394)
	at org.jetbrains.kotlin.gradle.tasks.AbstractKotlinCompile.executeImpl(Tasks.kt:335)
	at org.jetbrains.kotlin.gradle.tasks.AbstractKotlinCompile.execute(Tasks.kt:304)
	at java.base/jdk.internal.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at java.base/jdk.internal.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:77)
	at java.base/jdk.internal.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at java.base/java.lang.reflect.Method.invoke(Method.java:568)
	at org.gradle.internal.reflect.JavaMethod.invoke(JavaMethod.java:104)
	at org.gradle.api.internal.project.taskfactory.IncrementalTaskInputsTaskAction.doExecute(IncrementalTaskInputsTaskAction.java:47)
	at org.gradle.api.internal.project.taskfactory.StandardTaskAction.execute(StandardTaskAction.java:51)
	at org.gradle.api.internal.project.taskfactory.AbstractIncrementalTaskAction.execute(AbstractIncrementalTaskAction.java:25)
	at org.gradle.api.internal.project.taskfactory.StandardTaskAction.execute(StandardTaskAction.java:29)
[... 163 lines cut ...]
	at java.base/jdk.internal.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:77)
	at java.base/jdk.internal.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at java.base/java.lang.reflect.Method.invoke(Method.java:568)
	at java.rmi/sun.rmi.server.UnicastServerRef.dispatch(UnicastServerRef.java:360)
	at java.rmi/sun.rmi.transport.Transport$1.run(Transport.java:200)
	at java.rmi/sun.rmi.transport.Transport$1.run(Transport.java:197)
	at java.base/java.security.AccessController.doPrivileged(AccessController.java:712)
	at java.rmi/sun.rmi.transport.Transport.serviceCall(Transport.java:196)
	at java.rmi/sun.rmi.transport.tcp.TCPTransport.handleMessages(TCPTransport.java:587)
	at java.rmi/sun.rmi.transport.tcp.TCPTransport$ConnectionHandler.run0(TCPTransport.java:828)
	at java.rmi/sun.rmi.transport.tcp.TCPTransport$ConnectionHandler.lambda$run$0(TCPTransport.java:705)
	at java.base/java.security.AccessController.doPrivileged(AccessController.java:399)
	at java.rmi/sun.rmi.transport.tcp.TCPTransport$ConnectionHandler.run(TCPTransport.java:704)
	at java.base/java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1136)
	at java.base/java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:635)
	... 1 more
Caused by: java.lang.ExceptionInInitializerError: Exception java.lang.AssertionError: symbolic reference class is not accessible: class sun.nio.ch.FileChannelImpl, from class org.jetbrains.kotlin.com.intellij.util.io.FileChannelUtil (unnamed module @6069db50) [in thread "RMI TCP Connection(2)-127.0.0.1"]
	at org.jetbrains.kotlin.com.intellij.openapi.diagnostic.DefaultLogger.error(DefaultLogger.java:53)
	at org.jetbrains.kotlin.com.intellij.openapi.diagnostic.Logger.error(Logger.java:169)
	at org.jetbrains.kotlin.com.intellij.util.io.FileChannelUtil.setupUnInterruptibleHandle(FileChannelUtil.java:43)
	at org.jetbrains.kotlin.com.intellij.util.io.FileChannelUtil.<clinit>(FileChannelUtil.java:18)
	... 56 more

Could not connect to kotlin daemon. Using fallback strategy.
Caused by: java.lang.ExceptionInInitializerError: Exception java.lang.AssertionError: symbolic reference class is not accessible: class sun.nio.ch.FileChannelImpl, from class org.jetbrains.kotlin.com.intellij.util.io.FileChannelUtil (unnamed module @6069db50) [in thread "RMI TCP Connection(2)-127.0.0.1"]

stevebroshar, 2024-01-16:
Does compiles after upgrading gradle plugin per suggestion in popup. So... it's not really a suggestion. It's a requirement. 

Can't the sample be updated so consumers don't have to figure this out? Or at least put a LOUD note in the readme.

ghfab, 2024-10-18:
Hi, i just want to say i meet exaclty the same compilation issue using recent Android Studio (2024.1.2).
updgrading gradle does not solve the issue.
Is Camera2Basic still supported?