android/camera-samples #626: no View Bitmap  in debugger

aike1202, 2025-08-09:
[image]

BreezeAndMoon, 2025-08-12:
Android Studio Narwhal Feature Drop | 2025.1.2
Build #AI-251.26094.121.2512.13840223, built on July 26, 2025
Runtime version: 21.0.6+-13391695-b895.109 amd64
VM: OpenJDK 64-Bit Server VM by JetBrains s.r.o.
Toolkit: sun.awt.windows.WToolkit
Windows 11.0
Kotlin plugin: K2 mode
GC: G1 Young Generation, G1 Concurrent GC, G1 Old Generation
Memory: 4096M
Cores: 16
Registry:
  ide.instant.shutdown=false
  ide.balloon.shadow.size=0
  debugger.new.tool.window.layout=true
  ide.experimental.ui=true
  ide.images.show.chessboard=true
Non-Bundled Plugins:
  com.intellij.marketplace (251.26094.141)
  Dart (251.27623.5)
  jd-intellij (0.8.1)
  com.github.gitofleonardo.simplesqlitebrowser (1.0.7)
  com.nasller.CodeGlancePro (1.9.8)
  com.github.copilot (1.5.52-243)
  io.flutter (87.0)
  cn.yiiguxing.plugin.translate (3.6.2)

aike1202, 2025-08-12:
same version 
@BreezeAndMoon  wecaht:AiMi_0210

BreezeAndMoon, 2025-09-01:
update to Android Studio Narwhal Feature Drop | 2025.1.3 Canary 1
[image]